corrados/edrumulus #9: Circuit question

thijstriemstra, 2021-11-13:
hi, I'd like to give this a try but is this disclaimer still up to date or are there changes to the diagram?

[image]

corrados, 2021-11-14:
I am no hardware design expert. So, yes, the disclaimer is still valid. All I can say is that it works well for me.

Please note that this schematic does not show you all the needed components nor the input ports to the ESP32 are correct. There is just one schematic given for any piezo trigger and one for the hi-hat controller. If you want to use, e.g., 6 pads with stereo piezo support, you need to build the circuit for the piezo input 12 times and use the correct ESP32 input ports as given in
https://github.com/corrados/edrumulus/blob/10fa7ffbeec366b2b21db8af47cd8eb771b7b096/edrumulus.ino#L60
and
https://github.com/corrados/edrumulus/blob/10fa7ffbeec366b2b21db8af47cd8eb771b7b096/edrumulus.ino#L61

Please also note that Edrumulus is still just experimental software. It works ok but there is still a lot of room for improvement. Also, I only support the pads given in the enum
https://github.com/corrados/edrumulus/blob/10fa7ffbeec366b2b21db8af47cd8eb771b7b096/edrumulus.h#L26
If you have self-made pads, Edrumulus might not perform well at all.

thijstriemstra, 2021-11-14:
i would like to test with my roland cy-5, roland pd-5 and pd-125bk. looks like this project is worth a try so ordering components for this project today!

corrados, 2021-11-14:
Do you want to use the ESP32 or the Teensy? I would recommend the Teensy, actually.

thijstriemstra, 2021-11-14:
i have a esp32 devkit v4 i wanted to use but i can try teensy as well.

thijstriemstra, 2021-11-14:
ps. do you have the source of that fritzing sketch and could you add it to the repository?

corrados, 2021-11-14:
> do you have the source of that fritzing sketch and could you add it to the repository?

Here it is:
[edrumulus_new.zip](https://github.com/corrados/edrumulus/files/7534081/edrumulus_new.zip)

> and any advice on what type of piezo trigger/hhat control ports? A 1/4 stereo jack female pcb mount with 2 pins? or 3?

I am using normal stereo jack ports which are soldered to a small PCB as seen in the following picture: https://github.com/corrados/edrumulus/blob/main/algorithm/images/jackboard.jpg. These PCBs are taken from an old patchbay.

> I really like these: https://www.ebay.com/itm/324731292802

Wow, this looks really professional. Even Roland modules do only use the simply types.

> Also, is it possible to trigger this in a DAW using MIDI?

Edrumulus just sends out MIDI notes over serial to the host PC. If you are using Windows, you need the Hairless MIDI tool to convert it to a MIDI stream which a DAW can use.

In my [Linux script](https://github.com/corrados/edrumulus/blob/main/tools/run_edrumulus.sh), I am using the ttymidi tool and send the MIDI signal to [Drumgizmo](https://drumgizmo.org).

bzass, 2021-11-14:
> If you have self-made pads, Edrumulus might not perform well at all.

Hi, I'm also super impressed by your work - thanks for sharing this project and your active contributions! I would like to give it a try with ESP32 and a custom pad based on the Open Edrums project by Ryo Kosaka. I know that those are not currently supported. Would it be possible for you one day to write or record a short tutorial about your methodology of tuning parameters to the other pads or at least a practical explanation of parameters to allow for manual tuning? That would be much appreciated and make it possible for a broader community to tune them to their pads. Thanks once again!

corrados, 2021-11-14:
> Would it be possible for you one day to write or record a short tutorial about your methodology of tuning parameters to the other pads or at least a practical explanation of parameters to allow for manual tuning?

At present time it would be better that you record a wave file of your piezo trigger signal captured with a sampling rate of 8 kHz and hitting the pad according to the rules defined in https://github.com/corrados/edrumulus/tree/main/algorithm#test-signals. I could then add your wave file in the "signals" directory of the Git repo: https://github.com/corrados/edrumulus/tree/main/algorithm/signals. I could then tune the parameters for you.

> That would be much appreciated and make it possible for a broader community to tune them to their pads.

The problem is that the algorithms are not yet fix. I want to improve the positional sensing and also want to tune some other parts of the algorithm. Each change of the algorithm will most probably require the pad parameters to be adjusted. Therefore it does not make sense to fine-tune the parameters yet.

thijstriemstra, 2021-11-14:
Thanks for the feedback!

>  I would recommend the Teensy, actually.

Any reason? Faster? Cheaper?

> Wow, this looks really professional. Even Roland modules do only use the simply types.

Yea! Found [6 for $14](https://www.ebay.com/itm/124892424978) that I'll give a try.

corrados, 2021-11-14:
> Any reason? Faster? Cheaper?

The ESP32 has some unsolved hardware issues. The ADC multiplexer seems not to separate the inputs good enough. E.g., if I move the Hi-Hat pedal it changes the DC offset of another piezo trigger input which is bad. I did not see such issues with the Teensy. Also, the ADC spikes are much smaller for the Teensy.

bzass, 2021-11-14:
> The problem is that the algorithms are not yet fix. I want to improve the positional sensing and also want to tune some other parts of the algorithm. Each change of the algorithm will most probably [...]

Thanks a lot for your reply. Fully understood, I will wait for things to get more stable. It would be great to get some docs/tutorials after it happens so that we don't need to bother you with parameters for custom pads.

Thanks again for this great contribution.

thijstriemstra, 2021-11-14:
thanks, didn't know that. I'll get a teensy 4.0.
 
[image]

closed 2021-11-21 by corrados