corrados/edrumulus #40: Add version number

thijstriemstra, 2022-01-09:
It would be useful for people using edrumulus in their project to be able to display/determine a version for Edrumulus. For my first prototype, if I add a display, this would be the first thing I want to see when the device is on.

corrados, 2022-01-10:
This is now supported:
[image]

closed 2022-01-10 by corrados

thijstriemstra, 2022-01-11:
thanks @corrados Only I was more thinking of adding it to `edrumulus.h` which my customized version of `edrumulus.ino` loads. I would expect any user of edrumulus to copy the .ino, make their own adjustments, and include the external edrumulus library (which has the version nr). Your approach would be something the end user uses for their own version nr, e.g. `MyEdrums 0.1`.

thijstriemstra, 2022-01-18:
thanks. can you also use the version number in the changelog (and bump it up when a new changelog entry is added)? And creating a git tag after that would be great.

corrados, 2022-01-19:
Sure. It's done now.