corrados/edrumulus #86: Is Prototype 2 still supported/recommended? (get_prototype_pins)

ryanalexmartin, 2023-08-14:
In this method, there are a few prototype boards listed.  I hand soldered the Prototype 2 circuit, so I'm really hoping I don't need to solder the newer iterations.

If pin 5 is not set to low, then the code defaults to prototype 4.  If I am using the prototype 2 front end circuit, will the pin values and default settings for the prototype 4 board still work correctly?  If so, could this be reflected in a comment in the code?  If not, could prototype 2's PDF be removed, or a disclaimer added?

```
// -----------------------------------------------------------------------------
// ESP32 Dual Core -------------------------------------------------------------
// -----------------------------------------------------------------------------
#ifdef ESP_PLATFORM

int Edrumulus_hardware::get_prototype_pins ( int** analog_pins,
                                             int** analog_pins_rimshot,
                                             int*  number_pins,
                                             int*  status_LED_pin )
{
#ifdef CONFIG_IDF_TARGET_ESP32
  // Definition:
  // - Pin 5 is "input enabled, pull-up resistor" -> if read value is 1, we know that we have a
  //   legacy or custom board. Boards which support the identification set this pin to low.
  // - Pin 18, 19, 22, 23 define a 4 bit sequence which identifies the prototype hardware.
  // NOTE: avoid ESP32 GPIO 25/26 for piezo inputs since they are DAC pins which cause an incorrect DC offset
  //       estimation and DC offset drift which makes the spike cancellation algorithm not working correctly
  pinMode ( 5, INPUT );

  // check support of protoype board identification
  if ( digitalRead ( 5 ) == 0 )
  {
    // read the identification bit field and check the states
    pinMode ( 18, INPUT ); const int bit1 = digitalRead ( 18 );
    pinMode ( 19, INPUT ); const int bit2 = digitalRead ( 19 );
    pinMode ( 22, INPUT ); const int bit3 = digitalRead ( 22 );
    pinMode ( 23, INPUT ); const int bit4 = digitalRead ( 23 );

    if ( ( bit1 == 0 ) && ( bit2 == 0 ) && ( bit3 == 0 ) && ( bit4 == 0 ) )
    {
      // Prototype 5: 0, 0, 0, 0 -----------------------------------------------
      // analog pins setup:               snare | kick | hi-hat | hi-hat-ctrl | crash | tom1 | ride | tom2 | tom3      
      static int analog_pins5[]         = { 12,     2,     33,        4,         34,     15,    35,    27,    32 };
      static int analog_pins_rimshot5[] = { 14,    -1,     26,       -1,         36,     13,    25,    -1,    -1 };
      *analog_pins         = analog_pins5;
      *analog_pins_rimshot = analog_pins_rimshot5;
      *number_pins         = sizeof ( analog_pins5 ) / sizeof ( int );
      *status_LED_pin      = 21; // LED is connected to IO21 on prototype 5
      return 5;
    }
    else if ( ( bit1 > 0 ) && ( bit2 == 0 ) && ( bit3 == 0 ) && ( bit4 == 0 ) )
    {
      // Prototype 6: 1, 0, 0, 0 -----------------------------------------------
      // analog pins setup:               snare | kick | hi-hat | hi-hat-ctrl | crash | tom1 | ride | tom2 | tom3      
      static int analog_pins6[]         = { 36,    33,     32,       25,         34,     39,    27,    12,    15 };
      static int analog_pins_rimshot6[] = { 35,    -1,     26,       -1,         14,     -1,    13,    -1,    -1 };
      *analog_pins         = analog_pins6;
      *analog_pins_rimshot = analog_pins_rimshot6;
      *number_pins         = sizeof ( analog_pins6 ) / sizeof ( int );
      *status_LED_pin      = BOARD_LED_PIN;
      return 6;
    }
  }

  // if no GPIO prototype identification is available, we assume it is Prototype 4
  // analog pins setup:               snare | kick | hi-hat | hi-hat-ctrl | crash | tom1 | ride | tom2 | tom3  
  static int analog_pins4[]         = { 36,    33,     32,       25,         34,     39,    27,    12,    15 };
  static int analog_pins_rimshot4[] = { 35,    -1,     26,       -1,         14,     -1,    13,    -1,    -1 };
  *analog_pins         = analog_pins4;
  *analog_pins_rimshot = analog_pins_rimshot4;
  *number_pins         = sizeof ( analog_pins4 ) / sizeof ( int );
  *status_LED_pin      = BOARD_LED_PIN;
  return 4;
#else // CONFIG_IDF_TARGET_ESP32S3
  // analog pins setup:                 snare | kick | hi-hat | hi-hat-ctrl | crash | tom1 | ride | tom2 | tom3  
  static int analog_pins_s3[]         = {  4,     6,      7,        9,         10,     12,    13,    15,    16 };
  static int analog_pins_rimshot_s3[] = {  5,    -1,      8,       -1,         11,     -1,    14,    -1,    -1 };
  *analog_pins         = analog_pins_s3;
  *analog_pins_rimshot = analog_pins_rimshot_s3;
  *number_pins         = sizeof ( analog_pins_s3 ) / sizeof ( int );
  *status_LED_pin      = BOARD_LED_PIN;
  return 4;
#endif
}```

corrados, 2023-08-14:
That's correct. I just added some clarification in the source code to make it clear, see https://github.com/corrados/edrumulus/commit/6fbba963e4070379057935972f63135191c61021.

corrados, 2023-08-15:
> If so, could this be reflected in a comment in the code? If not, could prototype 2's PDF be removed, or a disclaimer added?

Since I changed the code and added the additional comment, I treat this issue as solved and will close it now.

closed 2023-08-15 by corrados