corrados/edrumulus #91: Floating signal on disconnected ports

3hhh, 2023-08-25:
The input and thus the midi signals generally start to float on ports without drums connected.
Obviously that's not great for testing, broken parts, performances, ...

The available solutions I'm currently aware of:
1. Use a simple pull-down resistor.
    Apparently there's some debate about whether this is advisable for analog inputs (e.g. [here](https://forum.pjrc.com/threads/23522-pull-up-down-and-analog-inputs)), but I guess I'll give it a try sooner or later.
1. Use e.g. cinch connectors with 3 pins where the third pin can be used to deactivate the input circuit ("off switch").
1. Fix it in software by detecting floating input signals and not generate midi output in that case.

Since the second option is not backwards compatible and also doesn't work with the Roland cable, I only consider the other two options viable at the moment.

corrados, 2023-08-26:
> The input and thus the midi signals generally start to float on ports without drums connected.

Does it make a difference if the pad is connected on that port or not? For my prototypes, I have a lot of unused ports but they do not behave differently if a pad is connected or not. Maybe your 100k resistor has long legs and acts like an antenna?

Usually, the power supply of the ESP32 board is responsible for the ghost MIDI notes. E.g., for my prototype 5 which get's its power directly from a raspberry pi and not from the PC USB port, the noise is much smaller and therefore the sensitivity is much better.

If you use a noisy power supply, I recommend to increase the threshold.

ocimjr, 2023-08-26:
Yesterday I was testing with a mac, plugged in a wall with 3 pins plug, and esp32 was plugged in a active usb hub... I was having a lot of spikes and it was impossible to use. I had to put threshold to 31.

Since you said the leg of a component, if it is long, can be an antenna, this can be my problem?
Or should I buy a dedicated power supply to esp32? With so, which one you suggest?

corrados, 2023-08-26:
> I had to put threshold to 31.

That doesn't sound correct at all. I never had it on such high values. Are these channels where you have to set the threshold to 31 are inputs which are not used? I assume that you have the correct resistors/diodes connected to all ESP32 ADC inputs which are used by ESP32. Is this the case?

ocimjr, 2023-08-26:
This only occurs when I have my mac plugged in a wall. I only use 4 inputs. I you don't my I put some photos of prototype

[image]
[image]
[image]
[image]

Can you check... I change the input on software to match the one on the board.

Thank you

corrados, 2023-08-26:
Have you changed the software to only use these 4 inputs?

ocimjr, 2023-08-26:
Yes -> 
Edrumulus.ino
`void preset_settings()
{

  // default MIDI note assignments
  edrumulus.set_midi_notes      ( 0, 38, 37 ); // snare
  edrumulus.set_midi_notes      ( 1, 36, 36 ); // kick
  //edrumulus.set_midi_notes      ( hihat_pad_idx, 42, 22 );
  //edrumulus.set_midi_notes_open ( hihat_pad_idx, 46, 26 );
  //edrumulus.set_midi_notes      ( hihatctrl_pad_idx, 44, 44 ); // Hi-Hat pedal hit
  //edrumulus.set_midi_notes      ( 4, 51, 53 ); // ride (edge: 59, bell: 53)
  //edrumulus.set_midi_notes      ( 5, 53, 59 ); // ride (edge: 59, bell: 53)

  // default drum kit setup
  edrumulus.set_pad_type ( 0, Edrumulus::PDX100 );  // snare
  edrumulus.set_pad_type ( 1, Edrumulus::PDX100 );  // snare
  //edrumulus.set_pad_type ( 1, Edrumulus::KD8 );  // kick
  //edrumulus.set_pad_type ( 2, Edrumulus::CY5 );  // Hi-Hat
  //edrumulus.set_pad_type ( 3, Edrumulus::FD8 );  // Hi-Hat-ctrl
  //edrumulus.set_pad_type ( 4, Edrumulus::CY8 );  // ride1
  //edrumulus.set_pad_type ( 5, Edrumulus::CY8 );  // ride2
  //edrumulus.set_coupled_pad_idx( 4, 5 );
}`

and in edrumulus_hardware.cpp

`Edrumulus_hardware::get_prototype_pins
// if no GPIO prototype identification is available, we assume it is Prototype 4
  // analog pins setup:               snare | kick | hi-hat | hi-hat-ctrl | ride1 | ride2   
  static int analog_pins4[]         = { 32 ,   34}; //33,     32,       25,            34,    12 };
  static int analog_pins_rimshot4[] = { 33 ,   35};//-1,     26,       -1,            27,    -1 };
`

corrados, 2023-08-26:
Ok, these software changes look ok.

> Or should I buy a dedicated power supply to esp32? With so, which one you suggest?

The problem is that you need the USB connection to the PC to receive the MIDI notes. In prototype 5, I use a serial connection to the Raspberry Pi and therefore can power the ESP32 differently.

As I mentioned in another thread, you might try to use a powered USB hub. Maybe the power supply of these hubs are a little bit less noisy.

ocimjr, 2023-08-26:
I'm already using a powered usb hub, I saw that thread. So, I'm running out of ideas.

ocimjr, 2023-08-26:
This is a plotter from ports 32 and 33. More in the region of 2000 and some spikes from 2100 and I event saw 2600

[image]

corrados, 2023-08-27:
These peaks look too high. I've never seen such high peaks on my prototypes. Maybe you have a cell-phone right next to your prototype or something else which does a lot of electrical interference. I can see that you have a slightly different ESP32 board than me. Mine has a bit less inputs. Maybe the power unit is different and makes more noise?

ocimjr, 2023-08-28:
No cellular is not near or other source of interference. 
I bought a expansion board that has a power supply. I should receive in 2 weeks time. I'll this will solve the issue. I update you when i received. Thanks again for your help.

3hhh, 2023-08-29:
> Does it make a difference if the pad is connected on that port or not? For my prototypes, I have a lot of unused ports but they do not behave differently if a pad is connected or not. Maybe your 100k resistor has long legs and acts like an antenna?

I haven't seen this issue yet neither and one of my tests indeed shows a zero signal without a pad connected. Now that I look at the prototype 2 diagram again I also see that a constant 1,65V should be at the ADC when there's no signal, i.e. practice confirmed theory.

I think I always misunderstood the manual sentence:
"If Edrumulus is installed on the micro controller and there is no analog front end circuit attached, you will get a lot of false triggering (i.e. a lot of MIDI notes). For the Edrumulus system to work correctly, all configured ADC inputs must have the analog front end circuit attached."

Apparently you meant the front end circuit without the pads...