corrados/edrumulus #93: Analog circuit

3hhh, 2023-09-11:
As already discussed, I'm pretty sure that the analog circuit is not optimal and likely doesn't work for everyone.
E.g. for me the proposed 22k voltage divider produced a rather bad SNR leading to many other issues.

The differences mostly come from the unknown Piezo output voltage range, which may be +- 0,1-10V.
The effective voltage range depends on:
- piezo hardware (size, thickness)
- trigger cushion foam (i.e. the maximum velocity at which a user can possibly hit the piezo)

In my particular case the voltage range of the piezo seems to be around +-1V and rarely more. This can be measured rather easily by plugging the piezo to the line in which runs at +-1V range. If there's no clipping, then you're apparently in that range.

I played a bit with the apparent industry-standard software [LTSpice](https://www.analog.com/en/design-center/design-tools-and-calculators/ltspice-simulator.html) to find a set of parameters that may work for me and interestingly found quite a different one from the proposed parameters.

Here's a comparison of the parameters proposed by edrumulus and the seemingly better ones for my case (+-1V, 2kHz spikes):
[image]
[image]
[simulation (needs to be renamed to edrumulus.asc)](https://github.com/corrados/edrumulus/files/12576561/edrumulus.txt)
(The C3=1p case is roughly identical to what edrumulus does = no capacitor.)

One can see that the voltage range in the edrumulus example is ~0,3V, whereas it could be ~2V as in the second example. Of course that also affects the SNR.

Therefore imho it would be more honest to tell people that they need to play with the parameters to find a circuit that may work for them.

Disclaimer: I still need to validate the simulation in practice.

corrados, 2023-09-11:
> If there's no clipping, then you're apparently in that range.

Some clipping is allowed. I have implemented a clipping compensation in software.

> Therefore imho it would be more honest to tell people that they need to play with the parameters to find a circuit that may work for them.

This is the current disclaimer on the main page:

**DISCLAIMER All these circuit diagrams are used for my first prototypes and work fine for me. Use them at your own risk.**

If you want to see a different text then please make a proposal.

corrados, 2023-09-11:
BTW, I may convert this Issue in a Discussion thread later on because as soon as I close it, most people will not find the investigation results you have given in your initial post. I assume that new users will only check for help in the Discussion forums.

3hhh, 2023-09-12:
> This is the current disclaimer on the main page:
> 
> **DISCLAIMER All these circuit diagrams are used for my first prototypes and work fine for me. Use them at your own risk.**
> 
> If you want to see a different text then please make a proposal.

I guess "In particular the circuits may not work for currently unsupported pieces of hardware." might be fair enough for a disclaimer.

However additionally some documentation on how to support new pieces of hardware would've been clarifying and helpful.

Currently the steps appear to be:

1. Start with prototype2 & optimize the analog circuit (atm it is unclear to me how to do that apart from random testing; I might know more once I can verify my simulation in practice).
2. Create samples for the edrumulus Octave model according to the existing doc. Probably getting them from the edrumulus hardware itself is more realisitic than from whatever sound card.
3. Analyze the samples in the Octave model to identify the edrumulus parameters. We discussed that in the forum.
4. Set the identified parameters in the edrumulus C++ code and test them in practice. Modify the available real-time parameters for hopefully better results. Possibly go back to previous steps.

corrados, 2023-09-12:
> I guess "In particular the circuits may not work for currently unsupported pieces of hardware." might be fair enough for a disclaimer.

done

3hhh, 2023-09-15:
Nice, thanks!

Btw I tested it in practice and unfortunately I couldn't record a signal that looked significantly better than previous ones (maybe there's some generic issue with the way I'm recording, idk). However I could lower the edrumulus threshold from an unusable 20 to around 10. Therefore I guess I'm going to stick with Rs=1k, Rp=100k, Cp=0 for the MPS 750x Tom for now. Cp dampens the gain as expected, but it doesn't look like I need it for the MPS 750x Tom.