corrados/edrumulus #97: spike cancellation cannot be set

3hhh, 2023-09-18:
[edrumulus_parameters.cpp](https://github.com/corrados/edrumulus/blob/f63b367f78c7ee1b8e91e68009a4e7668ca356ec/edrumulus_parameters.cpp) currently has no way of setting the spike cancellation parameter.

According to https://github.com/corrados/edrumulus/blob/f63b367f78c7ee1b8e91e68009a4e7668ca356ec/edrumulus.cpp#L31-L35 it should be set to 4 for ESP32 and to 0 for Teensy. However edrumulus_gui.py sets it to 4 when opened.

(I want to permanently set it to 2 for one of my pads.)

corrados, 2023-09-18:
> I want to permanently set it to 2 for one of my pads.

Spike cancellation is a global parameter and cannot be set for each pad separately.

> it should be set to 4 for ESP32 and to 0 for Teensy. However edrumulus_gui.py sets it to 4 when opened.

edrumulus_gui.py stores the current parameters in a settings file and recovers these settings on the next start. If you change the spike cancellation to 2, close edrumulus_gui.py and re-open it, is it then set to 2 or to 4?

corrados, 2023-09-18:
> Hm then maybe edrumulus_gui.py is misleading as it allows setting this per pad.

It's a bit of a hack. edrumulus_gui.py does not support global parameters. If you change the spike cancellation on one pad in edrumulus_gui.py, it will be changed on all other pads, too.

> Also it's a bit weird that it apparently overwrites the C++ define, which would be in place without edrumulus_gui.py starting.

This is the case with all user parameters. I am always using edrumulus_gui.py to manage all the trigger parameters.

> I also want to set the default because I'm pretty sure I'll forget about this setting after using it for a longer while...

Then you have to decide: Either use edrumulus_gui.py and let this tool manage all the parameters or do not use edrumulus_gui.py at all and only use hard-coded parameters in the Edrumulus firmware.

corrados, 2023-09-18:
> it should be set to 4 for ESP32 and to 0 for Teensy. However edrumulus_gui.py sets it to 4 when opened.

BTW: If you checkout the Edrumulus Git repo and did not change any parameter manually in edrumulus_gui.py, you are using my trigger settings which I am using for my current testing. This is the reason you are seeing a 4:

https://github.com/corrados/edrumulus/blob/6876846aedd3711f255a8a6c53989b148718fbea/tools/settings/trigger_settings.txt#L9

(110 is the MIDI note for the spike cancellation setting)

In this file you can also see that all pads have the same value, e.g.,
https://github.com/corrados/edrumulus/blob/6876846aedd3711f255a8a6c53989b148718fbea/tools/settings/trigger_settings.txt#L26
and
https://github.com/corrados/edrumulus/blob/6876846aedd3711f255a8a6c53989b148718fbea/tools/settings/trigger_settings.txt#L43

corrados, 2023-10-26:
I have now implemented an additional label for the "spike" parameter to make it clear that it is a global and not a pad-specific parameter:

[image]

3hhh, 2023-10-26:
Thanks, looks good to me! :-)

closed 2023-10-26 by 3hhh