corrados/edrumulus #100: Rim signals may be too strong for edrumulus

3hhh, 2023-09-29:
My MPS750x ride generates a rim piezo signal that is very strong. Setting the rim boost to 0 and the rim threshold to 31 still produces bad results (too many rim triggers).
It would therefore be good to have higher threshold or negative rim boost available.

Alternatively I could of course dampen the signal in hardware, but imho having a strong signal available shouldn't be a disadvantage.

By the way that's how one might distinguish bell, bow & rim on that ride: If the rim signal is weak (currently rim boost 0, rim threshold 31), it's probably the bell, if it's mediocre, it's the bow, otherwise the rim.
Possibly Millenium still does it differently though.

corrados, 2023-09-29:
> My MPS750x ride generates a rim piezo signal that is very strong

Are you shure that it is a piezo which detects the rim signal? I would have assumed that it is a switch and not a piezo. How have you setup your pad paramters? You can configure if it shall be a rim pizeo or rim switch.

3hhh, 2023-09-30:
I just updated the MPS750x branch so that you can see it for yourself.

corrados, 2023-09-30:
This is a recording of my PD-8 pad which definitely has a switch and no piezo on the second channel (first channel is the head piezo, second channel is the rim switch):
[image]
It looks very similar to what is in your file mps750x_ride_all:
[image]
I am very sure that you also have a switch and no pizeo on the rim.

corrados, 2023-09-30:
On test you could do is to measure the resistance of the rim if you have a multi-meter at hand. Then you can measure the Ohms when the pad is not touched and measure the Ohms when you press the choke (i.e., press the edge of the cymbal as if you would choke a real cymbal) which should close the rim switch.

3hhh, 2023-09-30:
Kk, I'll see whether I can do that.

Anyway [this line here](https://github.com/corrados/edrumulus/blob/b0b166b8f097994ed9c09867722e050e4969f20e/edrumulus.cpp#L502) only allows at most 10^((31-44)/10) = 5% as threshold.

corrados, 2023-09-30:
Yea, you are right. I think I have to fix this, soon. It's already too long on my [TODO list](https://github.com/corrados/edrumulus/blob/a12022450183326117c2aa946cc758a870b7e79f/doc/TODO.md?plain=1#L46) :-).

corrados, 2023-09-30:
Ok, TODO is done now: https://github.com/corrados/edrumulus/commit/55516d7516f6b2bb02411829039922a2fb868ff1

corrados, 2023-09-30:
> Anyway [this line here](https://github.com/corrados/edrumulus/blob/b0b166b8f097994ed9c09867722e050e4969f20e/edrumulus.cpp#L502) only allows at most 10^((31-44)/10) = 5% as threshold.

BTW, the line you referenced is for piezo rim triggers but you have to set your pad to rim switch which then uses the new formula I have just changed.

3hhh, 2023-10-02:
> On test you could do is to measure the resistance of the rim if you have a multi-meter at hand. Then you can measure the Ohms when the pad is not touched and measure the Ohms when you press the [...]

The resistance goes down from MOhm to 0 on the choke. I guess that indicates it's a switch?

corrados, 2023-10-02:
Yes, definitely. It's a switch.

3hhh, 2023-10-03:
Thanks for the update!

It kind of works now.
"Kind of" because it's less reliable than with the original module: E.g. if I hit the bell or bow with medium velocity, edrumulus will trigger a rimshot.

corrados, 2023-10-03:
I assume you are using the same resistors for the rim switch as for the piezos, right? Maybe it makes sense to play a bit with the resistor values to improve the rim switch detection.
In my setup with Roland pads, the same resistors work fine for piezo and switch but that may not be the case with your Millenium pads.

3hhh, 2023-10-04:
The signal and SNR is just fine, i.e. there's no need to play with the resitors at the moment. However I verified with my sound card that there is a rim signal whenever I hit the bell. That signal has roughly the same strength as the one I get whenever I hit the rim, which imho indicates that Millenium uses a different algorithm to identify rim, bow and bell. Hitting the bow also triggers the rim signal, albeit at a much lower level.

Anyway the original issue was fixed, i.e. this one can be closed.

closed 2023-10-04 by 3hhh