corrados/edrumulus #99: Disconnecting a rimshot input makes regular hits become rimshots

3hhh, 2023-09-29:
1. Use a pad with `pad_settings.rim_shot_is_used = true;` and possibly `pad_settings.rim_use_low_freq_bp = false;`.
2. Notice that everything works as intended.
3. Physically disconnect the rimshot input cable.
4. Notice that regular hits on the pad play the rimshot Midi note instead of the regular Midi note.

This may be a programming mistake or an instance of #91 on the rimshot input, idk.

corrados, 2023-09-29:
Have you tried to adjust the rim shot threshold to mitigate the incorrect rim shot detection on a disconnected cable?

3hhh, 2023-10-02:
> Have you tried to adjust the rim shot threshold to mitigate the incorrect rim shot detection on a disconnected cable?

Yes, but reliably only with a threshold of 31.

corrados, 2023-10-02:
Is it a rim switch or a piezo on the rim?

3hhh, 2023-10-02:
I tested it with the snare, i.e. pieo.

corrados, 2023-10-02:
Are you sure the piezo is directly connected to Edrumulus input? Because a piezo has a high resistance, see e.g. [this page](https://www.f3lix-tutorial.com/piezo-materials): "_A piezo material is basically an insulator. That means the internal Resistance is extremely high._".

I.e., if the piezo is directly connected, it should not make a difference if it is connected to Edrumulus or not.

corrados, 2023-10-02:
But anyway, this is actually not a supported use case for Edrumulus. You either connect your dual-trigger pad with a stereo jack connector to an Edrumulus input. Therefore, always both piezos or no piezo is connected. If you permanently want to disconnect the rim piezo, you simply configure the pad in Edrumulus as a single-trigger pad.

3hhh, 2023-10-02:
I don't want to permanently disconnect it. This is once again about accidents that may happen during playing.

The only good thing about this behaviour is that one probably notices it rather quickly.

A piezo that is not being hit is essentially a capacitor. That should explain why it makes slightly unsteady 1.65V from a power supply more steady.

3hhh, 2023-10-03:
So essentially you're saying that this is an instance of #91.

To prevent #91, only band pass filtering or some application logic such as "if there are more than X (configurable per port) triggers per second from a single port, disable that port & inform the user".

3hhh, 2023-10-04:
My signal and SNR seem very fine since I use Rp=100k, Rs=1k, velocity threshold 2 & high sensitivity. I could even remove the powered USB hub and power the Teensy from a desktop PC USB port without noticing any difference.

We digress though. If you don't want to fix that issue, that's fine as well. It's your call after all.

3hhh, 2023-10-11:
> My signal and SNR seem very fine since I use Rp=100k, Rs=1k, velocity threshold 2 & high sensitivity. I could even remove the powered USB hub and power the Teensy from a desktop PC USB port without noticing any difference.

Looks like I need to revise that as once I connect more than just the snare, I have a lot of noise in my circuit. Is this a common well-known issue or do I have to suspect that my circuit is bad somewhere?

corrados, 2023-10-11:
> I connect more than just the snare, I have a lot of noise in my circuit

I do not have such an issue with my ESP32 prototype even if I connect multiple pads to it.

3hhh, 2023-10-12:
So I kept the circuit & kick input pin stable (A0) and moved the snare pin around for testing on a Teensy 4.1.

The results are very confusing:
Snare on A9/A8 produces the aforementioned noise (but only with the kick connected), snare on A10/11 has less noise, same on A17/18 and snare on A14/15 (I tested https://github.com/corrados/edrumulus/pull/103) runs into ADC max (the LED blinks) on medium velocity hits. All of that just by recompiling edrumulus with new pin numbers and moving the snare input cable around.

Why does the ADC quality depend so much on the input pin? Any ideas what to try?
I'm lost.
Btw it's not the Teensy power input - I even ran it on batteries and experienced the aforementioned noise.

corrados, 2023-10-12:
> snare on A14/15 (I tested https://github.com/corrados/edrumulus/pull/103) runs into ADC max (the LED blinks) on medium velocity hits

That's a very interesting find. Maybe that is the reason why you were not happy with my analog front end design and selected your own resistor values. Maybe the reason was that your Teensy input was somehow attenuated causing the problems.

I have a Teensy 4.0 which works fine with the Edrumulus code. Obviously, there seem to be some incompatibilities between Teensy 4.0 and Teensy 4.1, unfortunately.

I can see in your code (https://github.com/corrados/edrumulus/pull/103) that you did changes at the ADC configuration. The "MIMXRT1062DVL6A keeper" gave me headaches when I was trying to support the Teensy. In your code I see that you use "INPUT_DISABLE". Do you really want to disable the inputs? Or is it to disable the keeper?

One reason for the effects you see could be (which I do not hope) that your hardware has a defect. Maybe some high spikes from the piezos did harm the inputs.

3hhh, 2023-10-12:
> That's a very interesting find. Maybe that is the reason why you were not happy with my analog front end design and selected your own resistor values. Maybe the reason was that your Teensy input was somehow attenuated causing the problems.

> One reason for the effects you see could be (which I do not hope) that your hardware has a defect. Maybe some high spikes from the piezos did harm the inputs.

Yes, maybe you're right that my Teensy is broken - especially since I noticed that I hadn't [minded the gap](https://www.youtube.com/shorts/QerwqYASMLg). ^^
So the Piezo may well have sent some negative voltages to the ADC input for a while.
On a side note: If that bricks the Teensy, I wonder whether the Teensy may brick when it's switched off and someone plays the drums for fun (e.g. kids)?
Anyway my parameters apparently even somehow work without the voltage divider lol.

> I can see in your code (https://github.com/corrados/edrumulus/pull/103) that you did changes at the ADC configuration. The "MIMXRT1062DVL6A keeper" gave me headaches when I was trying to support the [...]

It disables the keeper. I also removed the pin-specific code as the ADC library should do it better anyway. The ADC library e.g. also implements load balancing between the ADCs.

corrados, 2023-10-13:
> It disables the keeper. I also removed the pin-specific code as the ADC library should do it better anyway. The ADC library e.g. also implements load balancing between the ADCs.

More than two years ago, I have used the same functions as you now do but this was my comment: "try to disable keepers [...] seems not to work".
See https://github.com/corrados/edrumulus/commit/e66203f994f48930c29f1bcef8db23c44383e186.
And after that, I introduced the direct modification of the registers instead:
See https://github.com/corrados/edrumulus/commit/f52ad6018cc71ce0bc0395f0ca8d91faec11d891.

So, obviously, I was not happy with this function. I cannot remember the details but it certainly had a good reason to modify the registers directly because I also prefer to use the API functions normally.

3hhh, 2023-10-13:
Hm I now tested the Teensy with the voltage divider connected to several pins and some simple analogRead() code and the voltage continually degrades from 1.65V to 1.4V and possibly less. I'd say it's gone for good...

> > It disables the keeper. I also removed the pin-specific code as the ADC library should do it better anyway. The ADC library e.g. also implements load balancing between the ADCs.
> 
> More than two years ago, I have used the same functions as you now do but this was my comment: "try to disable keepers [...] seems not to work". See [e66203f](https://github.com/corrados/edrumulus/commit/e66203f994f48930c29f1bcef8db23c44383e186). And after that, I introduced the direct modification of the registers instead: See [f52ad60](https://github.com/corrados/edrumulus/commit/f52ad6018cc71ce0bc0395f0ca8d91faec11d891).
> 
> So, obviously, I was not happy with this function. I cannot remember the details but it certainly had a good reason to modify the registers directly because I also prefer to use the API functions normally.

I'm pretty sure you were unhappy because it didn't work as you attempted to use analog pin numbers with `pinMode()`, which only accepts real pin numbers.

corrados, 2023-10-14:
> I'm pretty sure you were unhappy because it didn't work as you attempted to use analog pin numbers with pinMode(), which only accepts real pin numbers. At least that's what the incorrect code you referenced indicates.

No, it was not about the pin numbers. That worked fine for me all the time. It really was about strange behavior of the ADC input.

Maybe your Teensy is not dead but just is not configured correctly and shows a weird behavior caused by this. Can you upload an official example source code from the Teensy page which makes use of the ADC to check if they are really broken?

3hhh, 2023-10-14:
> Maybe your Teensy is not dead but just is not configured correctly and shows a weird behavior caused by this. Can you upload an official example source code from the Teensy page which makes use of the ADC to check if they are really broken?

My tests were done with a slightly modified version of the official code from https://github.com/pedvide/ADC/blob/master/examples/analogRead/analogRead.ino

I'll also check the replacement I ordered with that code, i.e. I'll then know for sure.

corrados, 2023-10-15:
I recommend to start with my values of the resistors first since I use these schematics for a long time and very happy with it. It has a 100k resistor in series and that should also protect the Teensy/ESP32.

3hhh, 2023-10-15:
On 10/15/23 12:53, Volker Fischer wrote:
> I recommend to start with my values of the resistors first since I use these schematics for a long time and very happy with it. It has a 100k resistor in series and that should also protect the Teensy/ESP32.

Hm yes, my values are indeed somewhat more risky. However I believe that the clamping diodes are the real issue, not the series resistor:
The clamping diodes may need 0,6-0,7V to short, thus leaving the input at ~4V, but the Teensy ADC only supports -0,3V-3,6V. Your 100k series resistor mitigates the issue by almost always making the clamping diodes useless (the input voltage would need to be _really_ high to make the clamping diodes short). With 1k, I rely on them, but that may be a false sense of security as indicated above... A real solution wuld be another voltage divider to make the diodes short at 3,3V, but of course that's not great on a breadboard - possibly suitable for your PCB designs though.

On a plus side, my values should be a lot better at low piezo input voltages as you drop ~100/122 = 81% input signal voltage on your 100k series resistor whereas I only drop ~1/101 = 1% there. Possibly I could live with Rs=100k, Rp=100k and thus drop ~50% at Rs. Or choose Rp higher, but that produces other issues with the ADC, which needs low input impedance to be able to draw current quickly enough (one might be able to mitigate those issues with a capacitor I read, but oh well...).

corrados, 2023-10-15:
I was just curious if you are happy with my schematics if the Teensy is ok (and not broken) because I would like to know if the Millenium pads really have such a lower output voltage compared to the Roland pads.

closed 2023-10-20 by 3hhh