corrados/edrumulus #105: Hihat stomp is weak and cannot be mixed alone

3hhh, 2023-10-20:
The hihat stomp output was rather weak during my tests compared to the hihat hits.

This could be fixed by postprocessing it with a mixer; however edrumulus currently uses the same Midi note for a stomp as for a hit on the hihat.

Imho it would be better to use different notes, so that such volume issues can be fixed during postprocessing.
Also, stomping sounds slightly different from hitting a hihat in practice, i.e. one might want to use different samples anyway.

corrados, 2023-10-20:
There is a separate MIDI not for Hi-Hat stomp action, see https://github.com/corrados/edrumulus/blob/e8c524a2bbbc87cd74a47a6215480357342b2832/edrumulus.ino#L107

3hhh, 2023-10-21:
Thanks for the hint!

During my testing I had noticed 4 different notes for the hihat: Two for hits on bow and edge and another two for the open/closed bow/edge variant.
Maybe my stomps aren't detected?
I'll investigate further....

3hhh, 2023-10-22:
> Maybe my stomps aren't detected?

Yes, they weren't detected. I managed to fix it by setting a high threshold & sensitivity for the control port.

With a lower threshold I could observe my hihat pedal movements on the screen, which was really nice and would easily enable half-open hihat Midi notes, if one wanted to support that feature.

Anyway this particular issue is invalid.

closed 2023-10-22 by 3hhh