corrados/edrumulus #106: Pos sensing only works with middle piezo

3hhh, 2023-10-20:
The Millenium MPS750x snare has its piezo at the top of the pad.

This has the advantage of a reduced hot spot effect, but doesn't work so well with edrumulus pos sensing, which currently assumes the piezo to be located in the middle of the pad.

It would be nice to support pads with piezos at the top.

I believe a simple version might be possible by translating the current pos sensing as follows:
middle --> edge
edge --> edge
everything else (somewhere in between) --> middle / reduce distance to edge by 50%

I'll test that once I have some time.

corrados, 2023-10-20:
The PDX-8 has the piezo on the bottom of the pad. You can tweak the positional sensing to work of for that pad with the existing parameters pos_threshold and pos_sensitivity.

3hhh, 2023-10-21:
> The PDX-8 has the piezo on the bottom of the pad.

But in the middle, isn't it?

> You can tweak the positional sensing to work of for that pad with the existing parameters pos_threshold and pos_sensitivity.

Yes, I noticed that it produces some consistent values.
However obviously it believes the middle to be at the top.
It should be possible to adapt that behaviour though.

corrados, 2023-10-21:
>>    The PDX-8 has the piezo on the bottom of the pad.

> But in the middle, isn't it?

No, it is at the bottom.

> However obviously it believes the middle to be at the top.

No, it does not. I just outputs the distance to the piezo. More information you cannot extract from a single sensor.

ignotus666, 2023-10-21:
This was with a Megadrum, but I was able to get positional sensing working with a side-mounted trigger. The trick was to install a cone (no sensor) in the centre. You can see how it performed [here](https://youtube.com/watch?v=rB1VslJnXjE&si=jxRTz72flsqtkOcD).

corrados, 2023-10-21:
That's a nice trick, thanks for sharing.

3hhh, 2023-10-30:
> It just outputs the distance to the piezo. More information you cannot extract from a single sensor.

I then wonder how the PDX-8 behaves when you get close to the bottom. Does it correctly identify it as edge?
I guess not as that would contradict your distance statement.

So if a signle piezo can just measure the distance - which makes sense to me -, this problem cannot be solved in software only anyway (if you draw a circle on the top of a pad you'll notice that it hits both edges, but the middle as well - how should one decide then?) and I guess my current settings with the top piezo are as good as they can get.

corrados, 2023-10-30:
> I guess not as that would contradict your distance statement.

That's correct. If you hit at the bottom, it detects it as the center. You have to adjust the positional sensing parameters in a way that the actual strike in the middle position is detected as the middle position so that if you move to the bottom, the position does not change. Just if you move towards the upper edge, the position is detected as edge. This is the best you can do with a piezo which is not mounted in the middle (or you use the trick suggested by ignotus666 but this would require to change the hardware).

3hhh, 2023-10-30:
Ok, then this won't be implemented and can be closed.

closed 2023-10-30 by 3hhh