corrados/edrumulus #111: cymbal choke causes rim triggers

3hhh, 2023-11-06:
Using the cymbal choke currently causes low volume rim triggers (usually two or so).

Also, holding the choke sometimes seems to just delay playback rather than stopping it. The latter may be a drumgizmo issue though.

corrados, 2023-11-07:
Yes, this is true. I can see this with my Roland cymbal as well. But it is not noticeable during my normal playing.

The choke detection in Edrumulus is delayed a bit. Therefore, you will have a similar issue like you had with the cross-talk cancellation. The invalid detected strike will be there before the choke is detected.

Maybe you could try to suppress these MIDI notes with your Python script like you do with the cross-talk cancellation. Would be interesting to know if that works.

corrados, 2023-11-07:
I just tested this with my Roland TD-27 and the Roland cymbal pad and I sometimes see exactly the same behavior as in Edrumulus, i.e., a Note On with MIDI not 49 is output on pressing the pad (choking it):
[image]

3hhh, 2023-11-07:
Hm, I don't like to do things badly just because others do that, too.

I'll probably try to fix it with my cross-talk script.

corrados, 2023-11-07:
Would be interesting to know how your Millenium brain performs in this situation. Do you see similar effects?

3hhh, 2023-11-11:
This could partially be fixed via [my script](https://github.com/3hhh/xtalk). Only partially as the first incorrect MIDI event happens ~10ms before the choke, the second ~5ms and I only want to delay events by at most 5ms, i.e. the first will continue to pass.

3hhh, 2024-01-05:
Btw increasing the sense time helped with this issue. However it caused worse rim detection, so that I dropped the change again.

3hhh, 2025-05-27:
Nowadays I don't use the choke anymore.

I simply use soft MIDI notes (just move the pad a bit) after strong ones to choke.

Essentially this technique makes rim switches somewhat obsolete for choking.