corrados/edrumulus #112: hihat pedal cross-talk

3hhh, 2023-11-07:
If the hihat pedal is used, the outcome is usually not just a pedal MIDI note, but a regular medium velocity hihat note or even hihat rim note as well. This is obviously caused by the hihat cymbals moving during the stomp. The resulting notes should be suppressed. Using the cross talk setting on the hihat helps, but isn't great as the hihat is only meant to be suppressed in that particular case and not when played together with other instruments.

Related: #109

corrados, 2023-11-07:
I tested this with my Roland TD-27 and my VH-12 hi-hat and see exactly the same behavior as in Edrumulus:

[image]

You get a hi-hat stomp sound (MIDI note 44) but also multiple MIDI notes 46 which are the normal hi-hat hits.

ignotus666, 2023-11-07:
This is a common issue in electronic hi hats that use a hi hat stand and a bottom cymbal - as opposed to a single cymbal, no stand, and a separate foot controller. There are people who prefer to use the latter setup precisely to avoid this.

There isn't really an easy way around it, because if you supress note on messages while the hi hat is travelling up or down, what about an actual intentional hit? I don't want that supressed, and how is the module supposed to tell the difference? At best you could supress notes sent that coincide with the hi hat hitting the top and bottom ranges after moving, but again it's the same problem - what if I intentionally hit it at that moment?

The best I've been able to do is address it at hardware level, using soft foam on the bottom cymbal to prevent them from clapping hard together when stomped on, and some more down where the felt is at the pedal chain to soften the hit when it goes up.

corrados, 2023-11-08:
Thanks for you comment, ignotus666. Since you have a lot of experience with DIY e-drums, it would be highly appreciated if you could build your own Edrumulus hardware and give valuable feedback ;-). BTW, great to have you contributing to my new project. It was a pleasure to work with you on Jamulus :+1:.

ignotus666, 2023-11-08:
The pleasure was (and still is) mine!

I would like to get around to building an Edrumulus module, but before I do I need to know whether my kit would be supported - I can't seem to find a list of compatible pad types. A lot of work has gone into building the kit over the years and I'd be quite reluctant to scrap what I have:

- Piezo/piezo snare and toms. I'd say these are supported.
- Piezo/switch crashes. Not sure about this. Piezo gives a bow note and the edge switch an edge note + choke.
- Piezo/switch/switch hi hat and ride. Same as the crashes + bell switch. Here the edge switch goes through a 10k resistor to distinguish it from the bell switch (both switches go to ring on the jack so all three zones just use a single stereo jack, like Yamaha does with their pads).
- Hall-effect sensor + magnet for the hi hat controller. Needs a ~5v source. The Megadrum module measures voltage changes, not resistance as seems the case here.
- In total my kit uses 21 inputs + hi hat controller. Not sure if that many inputs are supported.

If any of these are not yet supported but there are plans to do so, I'd be happy to collaborate with testing. My pads are DIY but mimic the behaviour of existing commercial pads, so it wouldn't be adding support for just a single user. Except maybe the hall-effect hi hat controller, which I am a huge fan of, however. For literally less than €10 you can make a DIY controller that is virtually indestructable as there are no mechanical/moving parts, and it works really reliably.

3hhh, 2023-11-09:
On 11/7/23 22:59, ignotus wrote:
> The best I've been able to do is address it at hardware level, using soft foam on the bottom cymbal to prevent them from clapping hard together when stomped on, and some more down where the felt is at the pedal chain to soften the hit when it goes up.

Thanks for the hint!
I tried it and indeed it made the issue less prominent, but unfortunately didn't fix it entirely. I guess I'll still have a look into a software solution. I agree that it's suboptimal, but on the other hand false suppressions would only happen, if I wanted to play a 1/16 note on the hihat bow at 80bpm with 40% the velocity of the hihat stomp or so. This is pretty uncommon for my play style so far...

Regarding your question on edrumulus support, you can see the commercially available supported pads in the code (edrumulus_parameters.cpp). They include piezo/piezo snare & toms, piezo/switch crashes and a poti hihat. The list of supported inputs is currently limited to snare | kick | hi-hat | hi-hat-ctrl | crash | tom1 | ride | tom2 | tom3 (each with at most 2 channels), but imho it essentially depends on the Arduino hardware used. You could combine multiple to obtain more ports. 21 is quite an impressive number though...

ignotus666, 2023-11-09:
So, I see that 3-zone cymbals, the hall-effect hi hat controller and the higher number of inputs aren't supported. I'm afraid that's a deal-breaker for me. I imagine it could be relatively easy to support the cymbals and maybe even the hi hat controller, but combining several Arduino boards to get more inputs sounds a bit... too much hassle. I was hoping multiplexers could be used; I think I recall someone testing that with some success but I'm not sure to what extent.

3hhh, 2023-11-09:
On 11/9/23 10:36, ignotus wrote:
> So, I see that 3-zone cymbals, the hall-effect hi hat controller and the higher number of inputs aren't supported. I'm afraid that's a deal-breaker for me. I imagine it could be relatively easy to [...]

Hm the Teensy registers itself as Midi device via USB to whatever host you're using. So multiplexing those together shouldn't be hard - probably you can just use qjackctl and in the worst case it's a ~100 line script with python-rtmidi. But yes, you'd have to run edrumulus GUI two times at the moment.
The Teensy 4.1 has 18 analog input pins, i.e. it would support 18 mono inputs (e.g. one piezo trigger per pin) in theory. Edrumulus currently uses 1-2 per pad. Whether the Teensy CPU is powerful enough to keep up with an 18 pin load idk - possibly @corrados knows more.

ignotus666, 2023-11-09:
The problem I see with using more than one Teensy (or ESP or whatever) is crosstalk. Unless the boards are "aware" of each other, it would be an issue. I know [Edrumin](https://www.audiofront.net/eDRUMin.php) modules can be daisy-chained and crosstalk between all pads is taken care of, but I have no idea how feasible doing that is.

corrados, 2023-11-09:
> A lot of work has gone into building the kit over the years and I'd be quite reluctant to scrap what I have

Sure, that was my expectation. You have something which works and why to change it. I have to admit, my intention for you was to do something [like this](https://github.com/corrados/edrumulus/discussions/83) which is a cheap ESP32 dev board, a bread board and some components and build a simple single input prototype. This way you do not have to spend a lot of money and can test the edrumulus performance for dynamics and positional sensing. Then you can already give me very valuable feedback :-).

> Same as the crashes + bell switch. Here the edge switch goes through a 10k resistor to distinguish it from the bell switch

This is not supported in software. Maybe hardware changes are necessary but maybe we are lucky and only a software change is required.

> Hall-effect sensor + magnet for the hi hat controller. Needs a ~5v source. The Megadrum module measures voltage changes, not resistance as seems the case here

Edrumulus also just measures voltage changes. Of course, you will need a special front end for this but maybe the software can stay the same. This has to be tested by you since I do not have such hardware available.

> In total my kit uses 21 inputs + hi hat controller. Not sure if that many inputs are supported.

As I mentioned in another thread, this will be the most challenging thing. The original dual-core ESP32 chip is already at its limits for my 13 channel setup. This will definitely not work. But the core Edrumulus code is not bound to a special hardware. If we find a micro controller with enough ADC inputs, a fast CPU and sufficiently fast ADCs, this number of inputs should be possible. The core Edrumulus code is written to scale nicely. BTW, do you mean 21 pads or 21 channels (i.e., do you count a stereo pad as one input or two inputs)?

> I was hoping multiplexers could be used; I think I recall someone testing that with some success but I'm not sure to what extent.

Yes, [bbuster2 has done some work in this area](https://github.com/corrados/edrumulus/discussions/73#discussioncomment-5712821) but he did not respond to my latest question. At least we can conclude that there may be some possibility for multiplexers together with Edrumulus.

> Whether the Teensy CPU is powerful enough to keep up with an 18 pin load idk - possibly [@corrados](https://github.com/corrados) knows more

I don't know. I mainly use the ESP32 and have not done many profiling with the Teensy board.

3hhh, 2023-11-09:
On 11/9/23 17:49, ignotus wrote:
> The problem I see with using more than one Teensy (or ESP or whatever) is crosstalk. Unless the boards are "aware" of each other, it would be an issue. I know [...]

Ah, I currently work on a cross-talk script that works on MIDI level, i.e. as long as you mix the two MIDI devices together in front of my script, it won't matter. In fact it'll work with non-edrumulus MIDI devices, too.

ignotus666, 2023-11-09:
> my intention for you was to do something https://github.com/corrados/edrumulus/discussions/83 which is a cheap ESP32 dev board, a bread board and some components and build a simple single input prototype. This way you do not have to spend a lot of money and can test the edrumulus performance for dynamics and positional sensing. Then you can already give me very valuable feedback :-).

This sounds good to me - I've got one of [these](https://www.lilygo.cc/products/lilygo%C2%AE-ttgo-t-display-1-14-inch-lcd-esp32-control-board) lying around that someone gave me. Would this board work? Otherwise it's no problem, I see the ESP32 boards you use are very cheap, but it would save me waiting for one.

> Same as the crashes + bell switch. Here the edge switch goes through a 10k resistor to distinguish it from the bell switch
> 
> This is not supported in software. Maybe hardware changes are necessary but maybe we are lucky and only a software change is required.

All the inputs in the Megadrum have the same front-end circuit and support this, so it's certainly done in software there. The key is the 10k resistor on one of the switches, which goes in the cymbal itself.

> Edrumulus also just measures voltage changes. Of course, you will need a special front end for this but maybe the software can stay the same. This has to be tested by you since I do not have such hardware available.

This is great to know.

> BTW, do you mean 21 pads or 21 channels

I use 21 channels - 10 stereo inputs + 1 mono. My module has 56 inputs available in total so keeping the number down was never a concern ;)

> Yes, https://github.com/corrados/edrumulus/discussions/73#discussioncomment-5712821 but he did not respond to my latest question. At least we can conclude that there may be some possibility for multiplexers together with Edrumulus.

I'd be willing to give it a go. If they could be used it would make it so much easier to scale up the number of inputs.

corrados, 2023-11-10:
> This sounds good to me - I've got one of [these](https://www.lilygo.cc/products/lilygo%C2%AE-ttgo-t-display-1-14-inch-lcd-esp32-control-board) lying around that someone gave me. Would this board work?

I also have one of these. I just tried it out: Yes, it works :-).

> I use 21 channels - 10 stereo inputs + 1 mono.

Ok, then it's not so bad. Maybe a Teensy 4.0/4.1 is already capable of this number of channels.

3hhh, 2023-11-11:
To come back to the original issue:
I could fix this via https://github.com/3hhh/xtalk.

thijstriemstra, 2023-11-14:
>  Except maybe the hall-effect hi hat controller, which I am a huge fan of, however. For literally less than €10 you can make a DIY controller that is virtually indestructable as there are no mechanical/moving parts, and it works really reliably.

That sounds great @ignotus666, any resources on how to build such a diy version you can share?

ignotus666, 2023-11-14:
@thijstriemstra 

It's very simple. I use a A1302 hall effect sensor, but I think they've been out of production for a few years. You can find what I presume to be knock-offs on Aliexpress that work, but I've found them to be less sensitive than the original. However, [this](https://www.bitsboxuk.com/index.php?main_page=product_info&cPath=302_305&products_id=2084) one will work too.

These sensors have three pins: voltage in, signal out and ground - I'm not sure how this would connect to the edrumulus board, but I suppose you'd take a 3.3v source from the board and connect the sensor's signal out pin to the hi hat input. And gnd of course. Then you need a neodymium magnet. I use one that's 10 x 20 x 30 mm.

Place the sensor fixed to the frame under the hi hat pedal, and the magnet stuck to the underside of the pedal, in such a way that when closed, sensor and magnet are ~2-5 mm away from each other (not touching). The open/closed range depends on the power of the magnet and your preference - I use a range of about 25-30 mm. The sensor is sensitive to polarity, so before mounting everything, make sure you know which side of the magnet (and the sensor - narrower face up) is the correct one. I 3D printed parts for the sensor (so it's at the right height) and the magnet. The magnet holds itself in place as the part it attaches to under the pedal is metal. I'll post an image - it's a bit dark and dusty, but hopefully you get the idea.

[image]

To give you an idea, I've been using this sensor for almost 15 years. I used to use magnets recycled from a hard drive but "upgraded" that a few years ago. Controllers that use mechanical parts (pots, FSRs with rubber actuators, etc.) will eventually wear, degrade and break. Bar an accident, this one will keep going indefinitely ;)

corrados, 2023-11-14:
That's brilliant. I wonder if the voltage curve you get is direct proportional to the position of the pedal or is it more like a non-linear relationship? Anyway, any non-linear curve could be compensated for in software.

ignotus666, 2023-11-14:
It isn't linear but it's close. As you say, it's easily compensated for in software. I use Superior Drummer and assign the different open levels there.

thijstriemstra, 2023-11-14:
that's great @ignotus666. are the 3d print files on thingiverse or printables.com by any chance? 😄

ignotus666, 2023-11-14:
I had a go at publishing it on Thingiverse but I can't get it to work... so I'll attach the .stl files here. You might have to edit them to fit your hi hat stand, though it's such a simple design you could just use them as a reference to create your own. The hole at the top isn't really needed for the hall sensor - the magnetic field won't mind a plastic barrier, doh! :-)

[hall-effect_housing.zip](https://github.com/corrados/edrumulus/files/13357189/hall-effect_housing.zip)

thijstriemstra, 2023-11-15:
thanks! I'll give it a try and get back to you.

3hhh, 2024-01-05:
Btw I noticed that significantly increasing the scan time for the cymbals helped with this issue. However it also caused worse rim shot detection, so that I dropped the change again.

giff25, 2025-05-27:
This makes me think of defining different voltage Delta thresholds for the hall effect sensor. If its above a certain threshold then the likely intention is to stomp and any "notes" should be suppressed until the voltage becomes roughly constant or there is distinct indication the hi-hats are closed . Would this be computationally feasible?