corrados/edrumulus #115: Unable to run edrumulus_gui.py on Windows

thijstriemstra, 2023-11-08:
After following the manual and installing Python packages I tried running `edrumulus_gui.py` using:

```console
tools/edrumulus_gui.py rtmidi
```

which results in:

```
/usr/bin/env: ‘python3’: No such file or directory
```

I would suggest using this in the docs instead:

```console
python tools/edrumulus_gui.py rtmidi
```

but it results in:

```
Redirection is not supported.
```

Any idea?

thijstriemstra, 2023-11-08:
If I use Powershell instead:

```
 python .\tools\edrumulus_gui.py rtmidi
```

I get this error instead:

```
Traceback (most recent call last):
  File "C:\Users\foo\projects\edrumulus\edrumulus\tools\edrumulus_gui.py", line 593, in <module>
    midiin, port_name_in   = open_midiinput([s for s in rtmidi.MidiIn().get_ports() if in_name in s][0], client_name="EdrumulusGUI")
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\foo\projects\edrumulus\edrumulus\tools\edrumulus_gui.py", line 597, in <module>
    raise Exception("No Teensy Edrumulus device found")
Exception: No Teensy Edrumulus device found
```

But I'm using an ESP32 though..

corrados, 2023-11-08:
Did you run loopMIDI and HairlessMIDI before you started edrumulus_gui? See https://github.com/corrados/edrumulus/discussions/16#discussioncomment-5630286

corrados, 2023-11-08:
This help text should also be up to date: https://github.com/corrados/edrumulus/blob/main/doc/manual.md#setup-for-esp32-under-windows

corrados, 2023-11-08:
Also checkout the comment here: https://github.com/corrados/edrumulus/blob/main/doc/manual.md#edrumulus-user-manual

**"[Python](https://www.python.org/) to control Edrumulus trigger parameters in real-time.
    Install the Python PIP packages python-rtmidi and windows-curses."**

Did you install these Python packages?

thijstriemstra, 2023-11-08:
> Install the Python PIP packages python-rtmidi and windows-curses."

> Did you install these Python packages?

Yes, see my previous comment:

> Python 3.10.4 (tags/v3.10.4:9d38120, Mar 23 2022, 23:13:41) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import rtmidi
>>> rtmidi.API_WINDOWS_MM in rtmidi.get_compiled_api()
True
>>> midiout = rtmidi.MidiOut()
>>> midiout.get_ports()
['Microsoft GS Wavetable Synth 0']

> Did you run loopMIDI and HairlessMIDI before you started edrumulus_gui? See https://github.com/corrados/edrumulus/discussions/16#discussioncomment-5630286

No, but when I do start them, and then run `python tools/edrumulus_gui.py rtmidi`, I'm getting a new exception:

```
Traceback (most recent call last):
  File "C:\Users\foo\projects\edrumulus\edrumulus\tools\edrumulus_gui.py", line 37, in <module>
    from rtmidi.midiutil import open_midiinput
ModuleNotFoundError: No module named 'rtmidi.midiutil'
```

thijstriemstra, 2023-11-08:
With pr #117 I am able to run `python tools\edrumulus_gui.py rtmidi` but the prompt/window is just empty:

[image]

Hopefully this is because I haven't my edrumulus esp32 device connected (because I haven't flashed new firmware yet)? Have you/can you try to start the edrumulus gui without a device connected @corrados and do you also get a blank screen in that case?

corrados, 2023-11-08:
> >  Did you install these Python packages?

> Yes, see my previous comment:

You have to be very carefully with the Python packages. Unfortunately, there are multiple rtmidi packages available but only the python-rtmidi works with Edrumulus. Please make sure you actually installed exactly this package and no other.

> No, but when I do start them, and then run python tools\edrumulus_gui.py rtmidi, I'm getting a new exception:

> Traceback (most recent call last):
>  File "C:\Users\foo\projects\edrumulus\edrumulus\tools\edrumulus_gui.py", line 40, in <module>
>    import jack
> ModuleNotFoundError: No module named 'jack'

That indicates that the command line argument is not correctly given to the script. If rtmidi command line argument is given, no jack audio is used at all.

>  but the prompt/window is just empty [...] Hopefully this is because I haven't my edrumulus esp32 device connected 

Exactly. Edrumulus must be connected for the script to work. Otherwise I also get an empty screen.

thijstriemstra, 2023-11-08:
> Exactly. Edrumulus must be connected for the script to work. Otherwise I also get an empty screen.

Great, then I'm confident to upgrade firmware and have ability to tune the parameters. I'd hate to put my edrumulus out of commission and go through trouble of reverting the firmware. Could you merge #117 though, without it I'm getting path issues.

corrados, 2023-11-08:
Before you update, please tag your current Git commit to be able to easily go back to the old version (just in case ;-) ).

corrados, 2023-11-08:
> Could you merge https://github.com/corrados/edrumulus/pull/117 though, without it I'm getting path issues.

done

thijstriemstra, 2023-11-08:
thanks. I'll upgrade firmware tomorrow and if everything's fine, close this ticket.

thijstriemstra, 2023-11-10:
>  if everything's fine, close this ticket.

So yesterday we were all ready to test out the new code, flashed new firmware, started edrumulus_gui.py and unfortunately it resulted in same black command prompt.. no errors, nothing. How can I debug this issue?

corrados, 2023-11-10:
Have you checked that you have loopMIDI correctly configured with the names "EdrumulusIn" and "EdrumulusOut"? You can try to flip the assignments in HairlessMIDI. Maybe this is the issue.

corrados, 2023-11-10:
I have just tried it out with the latest Git main branch and it works fine for me:
[image]

corrados, 2023-11-10:
I am pretty sure that it does not make a difference but you could try to enter the tools directory and start the script from there (you are in the edrumulus root directory: [image]). At least this is what I usually do.

thijstriemstra, 2023-11-11:
> Have you checked that you have loopMIDI correctly configured with the names "EdrumulusIn" and "EdrumulusOut"?

Good point, I used a different name, let me try.

thijstriemstra, 2023-11-11:
>

Awesome, that worked:

> Good point, I used a different name, let me try.

[image]

I opened #121 to clarify things a little bit more, thanks for the screenshot.

closed 2023-11-11 by thijstriemstra

thijstriemstra, 2023-11-11:
> I am pretty sure that it does not make a difference but you could try to enter the tools directory and start the script from there 

I believe this potential issue was fixed with #117 but before that PR it would surely be an issue.