corrados/edrumulus #133: Bad cymbal rim dynamics

3hhh, 2023-12-17:
I noticed that the cymbal rim dynamics are generally bad - there's just one or two velocity levels even though I have rim boost set to zero.

This may be related to my experiments in https://github.com/corrados/edrumulus/discussions/126, but I doubt that as the cymbal piezo value should be taken for the MIDI velocity and not the rim switch value.

corrados, 2023-12-17:
I cannot reproduce this issue. With my Roland CY pad, I get good dynamics when I hit the rim.

3hhh, 2023-12-17:
Interesting. Then I'll have to investigate this further...

3hhh, 2023-12-19:
So I went to some lengths to obtain proper samples from my Teensy with [this code](https://gist.github.com/3hhh/94f7199d1ebea5dd0502415375400813).
(Btw it would be useful to have some standardized way of obtaining samples in the repo.)

The results:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Unfortunately I cannot see any fault in my signals as their peak voltage clearly indicates the velocity (IIRC my hard ride rim hit wasn't so hard). Also, the rim switch capacitor circuit correctly indicates that it was used - admittedly sometimes a bit earlier or later than the hit piezo.
So in total I still think that the issue must be with the edrumulus code...

Also, with regards to https://github.com/corrados/edrumulus/discussions/126 the plots look neither good for the sample counting of the rim switch open time nor for the second threshold method to discern the bell from the edge. :-(
Maybe the velocity of the hit also has to be taken into account somehow, idk. Or use the hit piezo ripple length, but I don't understand the electronic background for that atm and it makes the different rim input impedances pointless.
Maybe I'll go back to the original circuit...

[raw data download](https://github.com/corrados/edrumulus/files/13718414/all.tar.gz)

corrados, 2023-12-20:
Maybe it is an issue with the internal pad trigger parameters. Maybe Edrumulus detects the main peak at the beginning of the burst:
[image]
BTW, you should use logarithmic voltage-axis to better visualize the dynamics of the signal.

3hhh, 2023-12-20:
On 12/20/23 07:21, Volker Fischer wrote:
> Maybe it is an issue with the internal pad trigger parameters. Maybe Edrumulus detects the main peak at the beginning of the burst

Ok, I increased the scan time to 8ms and since then I obtain _some_ dynamics in the range of 70 to 127 on the MPS750x crash and ride. I still cannot obtain lower velocities though. In fact, if I move from the center of the pad to the rim at constant low velocity, I can get velocity 10-20 hits until I reach the rim - then it suddenly becomes a rim hit of velocity ~70. Any idea why?

The high scan time is only needed for rim hits - I guess I _sometimes_ (definitely not in my test above) hit the pad multiple times during a rim hit (once at the rim and possibly one more time at the regular piezo location - at least that way I appear to get better dynamics) and I guess those signals may come in at different points in time.

I haven't tested increasing the scan time with the Lemon hihat yet (it has the same issue).

3hhh, 2024-01-05:
I went back to the original voltage divider circuit with my parameters and I now have good cymbal rim dynamics for the MPS750x pads again. So apparently this issue was a byproduct of [my capacitor circuit for the rim switch](https://github.com/corrados/edrumulus/discussions/126).

The Lemon hihat is still not perfect imo, but I'll look into that separately. Maybe the parameters could be chosen a bit differently.

closed 2024-01-05 by 3hhh