corrados/edrumulus #145: Half-open hihat play tends to cause soft pedal stomps

3hhh, 2024-06-13:
Playing half-open on a hihat which uses a poti for the pedal is very likely to cause soft pedal stomps as the hihat hits cause the upper cymbal to be pressed on the poti, which in turn makes edrumulus think that the pedal was moved.

This is particularly annoying as it mutes the hihat and breaks any fluent play.

I observed this on my Lemon hihat.

corrados, 2024-06-13:
That's interesting. I am using a pedal with a poti and have not observed such an issue yet.

Have you already played with the threshold/sensitivity settings for the pedal stomp? I assume that the stomp velocity is very low if you strike with the stick. A higher threshold value may solve the issue.

As a workaround you could use your external cross-talk cancellation script. But I can see that you already committed a change in that repo :-)

3hhh, 2024-06-14:
Yes, you're right: A combination of the two seems to have solved the issue for now.

I was using a pedal threshold of 0. It worked just well except for this issue here. Luckily increasing threshold (6) and sensitivity (31) seems to have mitigated 90% of the issue whilst keeping the same pedal functionality as before. The remaining 10% of cases should be covered by my `xtalk` script.

3hhh, 2024-06-21:
Hm increasing the threshold now makes the pedal sometimes miss stomps.
It's somewhat hard to optimize the parameters for two different goals (get all pedal stomps, but also don't trigger them on half-open play)...
One part of the problem being that I had to increase sensitivity to max, i.e. pedal stomps usually come in at 127 or not at all (which makes it hard to suppress them).

Does edrumulus generate pedal notes, even if the pedal didn't reach 100%?
If so, changing that may fix my issue as it should block all stomps where the pedal wasn't fully pressed down.

corrados, 2024-06-21:
Maybe this definition makes your issues:

https://github.com/corrados/edrumulus/blob/2d845c52a174bd1504378c65cd1c027b53124c1d/pad.h#L210

I have defined a MIDI number of 100 to be the open/closed threshold. Maybe you can play with that value a bit and try to find a better value for you.

3hhh, 2024-07-04:
> Maybe this definition makes your issues:
> 
> https://github.com/corrados/edrumulus/blob/2d845c52a174bd1504378c65cd1c027b53124c1d/pad.h#L210
> 
> I have defined a MIDI number of 100 to be the open/closed threshold. Maybe you can play with that value a bit and try to find a better value for you.

Yes, `125` seems to work better.

I also adjusted the `Midimap.xml` to the below to obtain "more room" to play the various sounds (taking into account that the pedal is pressed down by hihat hits):

```
<?xml version='1.0' encoding='utf-8'?>
<midimap>
        <map note="36" instr="KDrumR" />
        <map note="38" instr="snare" />
        <map note="30" instr="snare_rimshot" />
        <map note="22" instr="hihat_closed" controlthresh="125" />
        <map note="26" instr="hihat_closed" controlthresh="125" />
        <map note="42" instr="hihat_closedtop" controlthresh="125" />
        <map note="46" instr="hihat_closedtop" controlthresh="125" />
        <map note="26" instr="hihat_open" controlthresh="0" />
        <map note="26" instr="hihat_open1" controlthresh="80" />
        <map note="26" instr="hihat_open2" controlthresh="30" />
        <map note="46" instr="hihat_opentop" controlthresh="0" />
        <map note="46" instr="hihat_open1top" controlthresh="80" />
        <map note="46" instr="hihat_open2top" controlthresh="30" />
        <map note="44" instr="hihat_foot" />
        <map note="48" instr="tom1" />
        <map note="50" instr="tom1" />
        <map note="45" instr="tom2" />
        <map note="47" instr="tom2" />
        <map note="43" instr="tom3" />
        <map note="58" instr="tom3" />
        <map note="55" instr="crash" />
        <map note="52" instr="crash" />
        <map note="49" instr="crash_top" />
        <map note="57" instr="crash_top" />
        <map note="51" instr="ride" />
        <map note="53" instr="ride_bell" />
        <map note="59" instr="ride_side" />
</midimap>
```

I'll test this for a while... If it ends up working consistently and you're interested, I'll submit a respective PR.